SwiCago/HeatPump #162: fan speed quiet and fan speed 1

mamrai1, 2020-08-30:
const byte FAN[6]              = {0x00,  0x01,   0x02, 0x03, 0x05, 0x06};
const char* FAN_MAP[6]         = {"AUTO", "QUIET", "1", "2", "3", "4"};
it seems that 0x01 and 0x02 reflect the same fan speed which in reality is quiet...

mamrai1, 2020-08-30:
I can confirm that fan speed 'quiet' byte '0x01' is wrong!
tried 0x04, 0x07, 0x0a but no joy...
Is there an easy way to find correct byte?
i tried on 2 units MUZ-GE71VA and MUZ-GE35VA

All other settings work as expected

SwiCago, 2020-08-31:
No clue, fan speeds are correct for my units and many other peoples. Maybe not yours, you'll have to experiment. Try using debug mode and watch debug packets being sent to you in a terminal. Use your IR remote and cycle through the speeds and see what that byte is being reported as. Should be easy enough to figure out. When you figure it out, I can make adjustments or notes about your unit type for others

mamrai1, 2020-08-31:
these are  the only different packets received  on fan quiet and fan 1:
QUIET
{"packetRecv":"fc 62 01 30 10 06 00 00 1a 01 00 00 00 00 00 00 00 00 00 00 00 3c "}

Fan 1
{"packetRecv":"fc 62 01 30 10 06 00 00 1b 01 00 00 00 00 00 00 00 00 00 00 00 3b "}

any clues?

mamrai1, 2020-08-31:
i have also seen this message but i can't decode:
{"packetSent":"fc 5a 01 30 02 ca 01 a8 "}
{"packetRecv":"fc 7a 01 30 01 00 54 ac 5c 03 10 40 3b 1a 00 00 00 00 00 00 f4 cc "}

SwiCago, 2020-08-31:
@mamrai1 
Packets that start with fc 62 01 30 10 02 are the ones that have fan settings returned.
The other 2 with fc 62 01 30 10 06 is compressor frequency, this has nothing to do with fan.
So debug again, this time set fan with your IR remote and wait a minute or two until you see a response start with fc 62 01 30 10 02
Watch for the fan data byte, which is located here
fc 62 01 30 10 02 00 00 01 03 04 **02** 00 00 00 0c 00 00 00 00 00 45

mamrai1, 2020-09-01:
Yes it's always 02 on fan 1 and fan quiet....

mamrai1, 2020-09-01:
Closing this there is no way around...my units MSZ-GE35VA (2012) and MSZ-GE71VA(2010) report 'quiet' fan as fan 1 (0x02) and doesn't recognise 0x01.

closed 2020-09-01 by mamrai1

SwiCago, 2020-09-04:
@mamrai1 if your units lowest speed is fan 1 =0x02, then maybe mod the lobrary and remove quiet from the list, so it can't report incorrectly for your case. It is open source, so you can play with it and change it to suit your units. Good luck

dzungpv, 2020-09-06:
@SwiCago i have the same problem, Quiet not support in my device, so it is quite=1 fan mode. I will change the code

mamrai1, 2020-09-06:
I use mitsubishi2mqtt so I removed the quiet mode from mqtt only...

dzungpv, 2020-09-06:
> 
> I use mitsubishi2mqtt so I removed the quiet mode from mqtt only...
> […](#)

If you use home assistant you can use my code: https://github.com/dzungpv/mitsubishi2MQTT

mamrai1, 2020-09-06:
I use gysmo38 code with some minor improvement.. Haven't removed the
'quiet' from code yet.

https://github.com/mamrai1/mitsubishi2MQTT

mamrai1, 2020-09-06:
> > I use mitsubishi2mqtt so I removed the quiet mode from mqtt only...
> > […](#)
> 
> If you use home assistant you can use my code: https://github.com/dzungpv/mitsubishi2MQTT I have option disable quiet mode on the web UI

Can please help adding your fix on my code because i use latest version from gysmo38 and added some fixes on webgui and also added widevane support on mqtt?

dzungpv, 2020-09-06:
> > > I use mitsubishi2mqtt so I removed the quiet mode from mqtt only...
> > > […](#)
> > 
> > 
> > If you use home assistant you can use my code: https://github.com/dzungpv/mitsubishi2MQTT I have option disable quiet mode on the web UI
> 
> Can please help adding your fix on my code because i use latest version from gysmo38 and added some fixes on webgui and also added widevane support on mqtt? (https://github.com/mamrai1/mitsubishi2MQTT)

Yes, i will merge the code when i have free time