SwiCago/HeatPump #166: Problems connecting with a Wemos D1 mini 3.0.0?

pertakman, 2020-11-30:
I’m struggling with a Wemos D1 mini 3.0.0 not being able to communicate with my Mitsubishi MSZ-FH35VE. I’ve followed the tutorials I can find I’ve tried 10 kOhm pullups as well as level converter, but no success. I read post by mgrondelaers that the CH340G on the Wemos D1 mini v3.0.0 pulls the line low or high permanently when the module is powered with 5V.

Is this the issue I’m having problems with? Are there any other boards out there with the convenience of USB adapter for easy programming where the Rx and Tx will work without this issue?

I’d really appreciate some guidance because I’m out of ideas on how to proceed.

Warmly, Per

unixko, 2020-12-01:
You have to Wemos D1 mini v2 or follow https://github.com/unixko/MitsuCon/issues/15 .

pertakman, 2020-12-01:
Thank you for the confirmation! A friend of mine let me try a D1 v2 and it worked perfectly.

I appreciate your rapid reply!

SwiCago, 2020-12-05:
Wemos are known to have problems. See gitter chat, many cases. I even have a note in my readme about those shit chips.

closed 2020-12-05 by SwiCago