SwiCago/HeatPump #168: Wrong LICENSE file (GPL3 vs LGPL3)

jite, 2021-02-03:
README and preamble in the header/src files state LGPL license, however the LICENSE file included in the root is the GPL3 license.

Here's a link to the correct license file: https://www.gnu.org/licenses/lgpl-3.0.txt

And Thanks for a great lib! Made a custom PCB (using ESP8266/ESP12F as MCU), it's been working great for a year or two 👍

SwiCago, 2021-02-06:
Updated, but to be honest who gives a s#$t. It is a free library lol. Plus git automatically added that link in the default readme when i created this project

closed 2021-02-06 by SwiCago

jite, 2021-02-06:
Heh, well there are some major differences between the two licenses with regards to how your free library affects stuff around it (copyleft).

Can’t say how many will actually care in this case, but in general it’s the users who care. Companies usually stay away from anything GPL/LPGL, and I personally prefer the MIT-license.