SwiCago/HeatPump #170: Unknown/invalid data returned

mcouture87, 2021-03-02:
Here is an instance where I get "garbage/unknown" data:

{"temperature":73,"fan":"4","vane":"","wideVane":"|","mode":"���?�E @���?"}

I'm using https://github.com/gysmo38/mitsubishi2MQTT  that uses SwiCago/HeatPump libraries, so I am assuming there might be some issue at this level.

Anyone have any suggestions?

mcouture87, 2021-03-02:
Looking at MQTT Explorer's history, I see another 'garbage string' that just happened a few minutes ago...

When it happens, then HA reports the head unit's temperature at 79* until the next poll/update of the unit...

{"temperature":71,"fan":"AUTO","vane":"3","wideVane":"|","mode":"���?E @���?"}

SwiCago, 2021-03-06:
Open an issue on their git. Not mine. If you use the examples in this git and get these problems, then it could be my library. Doubt it though, as the library base level has been the same for years.

closed 2021-03-06 by SwiCago