SwiCago/HeatPump #174: Ethernet connection instead WiFi

gfreccia, 2021-09-02:
Hi All,

I want to use Ethernet connection instead WiFi (I've Arduino Mega 2560 + W5100 Ethernet Shield).
Do you have any suggestion about how to modify the code?

Thank you!

SwiCago, 2021-09-02:
Read this and you should be able to change the demo code to support ethernet. The library itself has nothing to do with ethernet or wifi

https://create.arduino.cc/projecthub/bjnhur/simple-mqtt-message-board-with-arduino-ethernet-73672d

closed 2021-09-02 by SwiCago

gfreccia, 2021-09-03:
Hi!

Thanks for the answer.

I've already Arduino Mega 2560 + W5100 Ethernet Shield working for another project (MQTT).
So I think that I can adapt _mitsubishi_heatpump_mqtt_esp8266_esp32.ino_, moving it to my hardware.
Is it the right way?