SwiCago/HeatPump #175: possible bug? mqtt not connecting when built with esp8266 (esp01) 3.0.2

logon84, 2021-11-30:
Hi. I always use the mqtt code example from the repo as base to communicate to the heatpump using an ESP01, and the setup has been working for 3 years. So today, after updating Arduino IDE (clean install), boards, and libraries, and then programming the esp01, the heatpump device was not talking to the mqtt broker. It took me hours to determine that the mqtt code compiles ok  but ESP01 is unable to connect to mqtt broker (using board data v.3.0.2). It seems to me that ESP01 connects to wifi and when it tries to connect to mqtt, suddenly it reboots. Then the wifi reconnection starts again and again. 

If you downgrade to board data to 2.7.3 in Arduino IDE then the code compiles and works as it should. Can someone also test this?

SwiCago, 2021-12-01:
Sounds more like an issue with the MQTT library and not the HP library or could be that your ESP is losing connection with WIFI and then trying MQTT loop. If you lose wifi and do MQTT loo, this could cause a reboot.
Try adding a check if WiFi.status == WL_CONNECTED before doing MQTT connect check in loop. If not connected, have it refresh it's WIFI or wait till wifi back and then do MQTT connect check. Don't do MQTT connect check/loop if Wifi is not connected.

Here is a page that shows something that might work
https://randomnerdtutorials.com/solved-reconnect-esp8266-nodemcu-to-wifi/

The sample provided in this library is just that, an example. To make it 100% robust for your network, you need to make changes or use one of the 3rd party alternative firmwares that imports this library, such as tasmoto or esphome.

logon84, 2021-12-02:
> Sounds more like an issue with the MQTT library and not the HP library or could be that your ESP is losing connection with WIFI and then trying MQTT loop. If you lose wifi and do MQTT loo, this [...]

Thanks SwiCago, I will try to switch to Tasmota then. Thx

closed 2021-12-03 by SwiCago