SwiCago/HeatPump #178: generic question about X105 using

PatAlf, 2022-02-15:
Dear all,
I understand that this projetct has reproduced the packet structure send initialy by a remote control connected on the X105 by using a wifi ESP8266 on the X105.

Another project (https://github.com/vilppuvuorinen/pymelcloud) is doing the same kind of pump control, but directly throught the native wifi interface connected on the X105.

Is there a chance to be able to complete this project with some features already implemented into yours ?
I was thinking on Service setting for example (be able to modify the paramter that define the fan behavior once the temperature is reached).

Thanks for your comments.

Just to present myself, I'm new on this kind of home assistant (whatever system we are talking about), and try to find out the best way for me to interacting with my MFZ-KT50 + Wifi adapter (Melcloud).
First subject : using a remote temperature sensor instead of the one integrated
Second subject : be able to activate or de-activate the fan behavior once the temperature is reached)

I've found some documentation that explain how I can physicaly modifiy this behavior by cutting a jump resistance on the mainboard.... this is something I would avoid.

SwiCago, 2022-02-15:
This library uses cn105, not x105. It is not based on a remote, but rather packets sent from a mitsubishi controler mhk1 or melcloud device.
This library does not piggy back off any online services, like the git you posted.
Remote temperature already exists, just look at functions available for library. 
As for special functions, you can use custom packets, if you know what to send for your specific unit..
Use the chat linked at the top of the readme of this project for questions. Issues should only be opened for bugs,  not questions

closed 2022-02-15 by SwiCago

PatAlf, 2022-02-16:
> Android [4].
> Links:

Thanks for the anwser.
Sorry to have post my question on the bad area, this is the first time 
for my as Github's guys.
I've made a mystake, about the connector, it's really the CN105, where 
in fact all the Mitsubishi accesory are connected to.

I've of course understand that your code send packets that have been 
initialy sniffed between a remote control on the CN105 and a mitsubishi 
system.
The other git i was refering to was in fact doing the same kind of job, 
but by connecting to MelCloud in order to send orders through the 
mitsubishi wifi adatper plugged on the CN105?

One question is , can we use your code in order to send localy the 
packets through the mitsubishi wifi adapter , like you do with the 
ESP8266 ?
I'm right if i tell that the ESP8266 only reating a remote connection to 
the CN105, like the mitsubishi does for the Melcloud service ?

I'm trying to understand the logic between the two ways of sending 
orders.
Yours send MQTT requests to CN105.
Pymelcloud send order to a remote service (MelCloud) that at the end use 
the CN105.

I do not need to have a feedback to any kind of other software or home 
assistant right now.
Just be able to try to adapt your code and see if i can use it localy 
with the Mitsubishi wifi adapter i've got already for MelCloud.

One issue of course is that programming under C or Python is new for me 
and creating of course the first troubles when trying to understand the 
codes.
But, i'm some experiences in other languages and TCP/IP sniffing with 
Wireshark.

Thanks if you can give me some answers.

Patlaf