SwiCago/HeatPump #183: Is there a way to make the air conditioner use an external temperature sensor?

amayii0, 2022-09-01:
(moving out of other topic) 

Is there a way to make the air conditioner use an external temperature sensor?

_Originally posted by @AntoxaSa in https://github.com/SwiCago/HeatPump/issues/13#issuecomment-1233136907_

fastbike, 2022-09-01:
It depends on your system. If there's no direct connection, then use an external sensor with a control system like Home Assistant. That will be a separate node.
My unit has a sensor built into the indoor unit, and another on a wired control panel in the same room. I find the control panel one to be accurate enough.

AntoxaSa, 2022-09-01:
I use the home assistant, the thermostat is automatically thrown from the air conditioner there. The sensor prokidyvaetsya internal air conditioner. And how to use an external temperature sensor for example akara?

jb123567, 2022-09-02:
I'm using this library in ESPHome which within Home Assistant allows you to use data from an external sensor to override the internal thermostat.

AntoxaSa, 2022-09-07:
> I'm using this library in ESPHome which within Home Assistant allows you to use data from an external sensor to override the internal thermostat.
is there an example implementation through esphome?

SwiCago, 2022-09-12:
yes, use command setRemoteTemperature(float TempInC)

closed 2022-09-12 by SwiCago