SwiCago/HeatPump #186: Missing supported device MSZ-AP20VG

Sprinterfreak, 2022-11-23:
Voided my warranty today on MSZ-AP20VGK using an ESP-WROOM-32 module from az-delivery. It does work now.

The unit is factory equipped with MELCloud Wifi adapter which of course did not work from the beginning. (waste of money) They seem to have the conceptual defect of requiring an internet connection. Would be glad to see Mitsubishi fix that. :D

Sprinterfreak, 2022-11-23:
So in the meantime I still can confirm it working but with minor issues.
I use this library via [gysmo38](https://github.com/gysmo38) /
[mitsubishi2MQTT](https://github.com/gysmo38/mitsubishi2MQTT) and have seen some issues.

  * can not set temperature below 17°C, seems to crash AC, temp jumps back to 23°C and sometimes unit behaves unexpected. Vanes start to move or for some time refuses further commands.
  * vane and fan control changes do not always go through
  * mode only fans made the unit blink it's error led.

Most important functions like on/off, temp between 17-22°C, AUTO/COOL/HEAT do work reliably so far.

sethkinast, 2022-11-23:
Working device reports go in https://github.com/SwiCago/HeatPump/issues/13

closed 2022-11-24 by Sprinterfreak