SwiCago/HeatPump #189: Add to Compatibility List

redstorm1, 2023-02-12:
Confirmed working with 
MSZ-LN18-60VG Series, New Zealand Black Diamond High Wall Units.
PUZ-ZM125 Ducted Unit

I used a WT32-ETH01 (Wireless Tag) however UART2 was non responsive and had to use UART0 and disable logging like a NodeMCU.

Not sure why UART2 was non functional on this ESP...

sumnerboy12, 2023-02-13:
Hey @redstorm1 - fellow Kiwi here, looking to use the WT32-ETH01 as well for my Mitsi AP50 - would you mind sharing your code?

redstorm1, 2023-02-13:
esphome:
  name: mitsi

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16
  
  manual_ip:
    static_ip: 192.168.1.xxx
    gateway: 192.168.1.x
    subnet: 255.255.255.0
    dns1: 192.168.1.x

external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump

climate:
  - platform: mitsubishi_heatpump
    name: "Guest Heat Pump"
    hardware_uart: UART0
    supports:
      mode: [HEAT_COOL, COOL, HEAT, FAN_ONLY]
      fan_mode: [AUTO, LOW, MEDIUM, HIGH]
      swing_mode: ["OFF", "VERTICAL"]
    visual:
      min_temperature: 16
      max_temperature: 31
      temperature_step: 1.0

redstorm1, 2023-02-13:
Don't bother with pullup resistors just wire straight to tx/rx as per the diagram

sumnerboy12, 2023-02-17:
> I used a WT32-ETH01 (Wireless Tag) however UART2 was non responsive and had to use UART0 and disable logging like a NodeMCU.

Did you try `Serial1`?

redstorm1, 2023-02-20:
> > I used a WT32-ETH01 (Wireless Tag) however UART2 was non responsive and had to use UART0 and disable logging like a NodeMCU.
> 
> Did you try `Serial1`?

UART2, UART1, and tried baud rates 1200 -- 115200 and all in between.

SwiCago, 2023-03-27:
Please use #13vto add your units and if anything was required to make it work.

closed 2023-03-27 by SwiCago