SwiCago/HeatPump #190: Add timeout to remote temperature readings

mbbush, 2023-02-23:
There is a scenario that can happen when using the setRemoteTemperature method on this library, where, if the data source for remote temperature goes away and stops sending updates, the heat pump just stays on, doing whatever it was doing at the time of last update (heating, cooling, idle, etc), forever. 

I implemented [a fix](https://github.com/gysmo38/mitsubishi2MQTT/pull/201) for this [issue](https://github.com/gysmo38/mitsubishi2MQTT/issues/167) in one of the projects using this library, which adds a check which runs in the main event loop, for whether it's been longer than the configured timeout since the last remote temperature value was received, and if so, reverts to internal temperature. 

My question is: is this functionality that would be useful/appropriate to put in the library itself? It seems like most if not all projects which use external temperature would welcome this change, and it seems to fall in the category of "related directly to the heat pump and its mitsubishi-defined protocol" rather than any particular home automation integration. The main argument I can see for not putting it in the library would be the performance overhead of tracking it. I don't know enough about C or development for embedded devices to know how much of a problem that would pose. 

What do you think?

Side note: I'd love to be able to still see the reading from the internal temperature sensor even when remote temp is active, but as far as I can tell from a non-expert look at the serial protocol debug output, the unit just doesn't expose it.

sumnerboy12, 2023-02-23:
I have also built this logic into my [firmware](https://github.com/sumnerboy12/OXRS-BJ-HeatpumpController-ESP-FW). I am of the opinion that it is better to keep this sort of business logic out of low level libraries wherever possible. But just my 2c.

SwiCago, 2023-03-27:
The mkh11 does the logic on its own as well. So it was kept out. If you want to use remote temperature in your arduino main loop, then you should add logic to reset or do what ever other defaults you want when you no longer get a signal from your mqtt server.

closed 2023-03-27 by SwiCago