SwiCago/HeatPump #191: Compatible with MSZ-AY indoor units?

ThiefMaster, 2023-03-22:
I just received a quote from a company to install AC at my place, and they propose  `MSZ-AY25VGK` units. The `MSZ-AY` line seems to be pretty new and seem to have wifi built-in, but according to [this site](https://confort.mitsubishielectric.fr/entreprise/nos-produits/pompe-chaleur-residentiel-airair/unite-interieure-multi-split-mural-compact-1) CN105 is also available.

The company also recommends installing wired remote controls for each unit.

Does anyone happen to know if:

- these models are compatible with this library;
- the CN105 port is used by the wired remote control, ie if i'd have to choose between making the units smart in a non-crappy way and using the wired remote control?

SwiCago, 2023-03-27:
Current units use cn105 for either wifi(mel/kumo cloud) or hard wired mhk11 units. So I would assume you could add this wifi library to your cn105 and chances are it will work. Only one way to find out, try it out.

closed 2023-03-27 by SwiCago

apppie123, 2023-11-24:
> I just received a quote from a company to install AC at my place, and they propose `MSZ-AY25VGK` units. The `MSZ-AY` line seems to be pretty new and seem to have wifi built-in, but according to [...]

Did you try it already?
I have a module that doesn't seem to work on a MSZ-AY. I'm not sure if the problem is the AY unit or the module. I don't have access to it at the moment so I'm curious with your experience.

ThiefMaster, 2023-11-24:
In the end I went for a different company and Daikin units, so unfortunately I'm not going to be able to try it.