SwiCago/HeatPump #193: What kind of connector

Manhem, 2023-04-22:
Hi.
Does anyone know what kind of connector we find on the WiFi module and where to find it?
Maybe Digikey part number.
I don't want to dismount my air heatpump, still warranty on it.
It has been much easier just to replace the WiFi module with my "own".
No need to distroy something and I can put the new one in the same place as the old WiFi module.

fastbike, 2023-04-22:
Did you read https://chrdavis.github.io/hacking-a-mitsubishi-heat-pump-Part-1/

brgaulin, 2023-04-23:
From the readme, there is a link to the digikey website here: https://github.com/SwiCago/HeatPump/blob/master/README.md?plain=1#L138

From there, the datasheet has alternative connectors if you want a different kind. I am in the middle of trying this out myself, and went with `PALR-05VF` + `APALPAL22K305`. That should enable me to make an "extension cable" that interfaces with the cable going to my existing kumo device. The wires will then get soldered directly into my ESP8266 with strain relief on a 3d printed box.

If you want the solderable connectors, the datasheet on digikey shows many options. I think the one you would want to match the kumo devices is a `S05B-PASK-2`

Manhem, 2023-04-24:
That connector is for the CN105 connector.
I would like to have that one you can find in the WiFi module and that is not the same.

brgaulin, 2023-04-24:
The CN105 is the most common type. If you don't have that one, can you share what you are trying to mimic? Pictures or model numbers would help. I only know the CN105, but hopefully someone else can help with that extra context.

Manhem, 2023-04-26:
The module I will connect to is a WiFi interface, model: MAC-577IF2E.
Here are some pictures. 
I apologize for the poor quality of one of the pictures.
So here you can see what kind of connecter I like to have.

[image]
[image]
[image]
[image]

brepdcom, 2023-05-04:
If you can measure the pin pitch, you can narrow down the possibilities significantly.  Start looking at JST's Wire-to-Board connectors with the pitch known and the part in hand, and you can probably figure out the specification.  
https://www.jst.com/products/crimp-style-connectors-wire-to-board-type/

Igor-Yavych, 2023-05-23:
You won't damage warranty because parts you need to remove to install it are mostly the same parts you need to remove for installation of the unit itself

SwiCago, 2023-06-16:
Chat link is fixed https://app.gitter.im/#/room/#Mitsubishi-Heat-Pump_Lobby:gitter.im
This question is beyond the scope of this library. Please ask in chat if anyone can help source parts

closed 2023-06-16 by SwiCago