SwiCago/HeatPump #194: Homekit support?

Olivierbkk, 2023-05-01:
Hi,

Anyone managed to make this work with apple homekit?

mayankst, 2023-05-17:
It works fine through home assistant integration, exposing it to the HomeKit.

Olivierbkk, 2023-05-18:
Thanks,

I am trying to stay away from home assistant at least for now as I like to keep things simple... I am using ESP32s with HomeSpan to make my own devices, I was wondering if there was a way to get this working with homespan 

[https://github.com/HomeSpan/HomeSpan](https://github.com/HomeSpan/HomeSpan)

brgaulin, 2023-06-11:
I think you have this issue on the wrong project. For questions on how to extend HomeSpan, I think it would be best to open an issue within HomeSpan.

If you are encountering a bug that related to this project though, please do let us know what that is and we can try to help.

An example of a similar integration can be found [here](https://github.com/geoffdavis/esphome-mitsubishiheatpump). ESPHome is a similar firmware to HomeSpan, but does not currently support HomeKit/Matter to my knowledge.

SwiCago, 2023-06-16:
Ask questions in chat, maybe someone already solved that
This project is only for the library to control the units
https://app.gitter.im/#/room/#Mitsubishi-Heat-Pump_Lobby:gitter.im

closed 2023-06-16 by SwiCago