SwiCago/HeatPump #197: gitter unavailable

wBrhy2, 2023-06-02:
Hello, I wanted to enter the chat to ask for help, but it's no longer active.

I have a MSZ-FH15NA and I am using an ESP32. I have tried all UARTs I can 

- UART0 logging disabled (baud at 0) and the CN105 on UART0
- UART0 logging disabled, disabled and CN105 on UART2
- UART0 enabled and CN105 on UART2
UART1 is not exposed on my board) and if I try baud 2400 (which has worked for some), I can no longer access my ESP32 at all. I can use the interface just fine, but nothing actually happens. I am using pull-ups (10k) from RX and TX to 5v as well as 20k from RX and TX to GND to serve as a level shifter. I have also tried no pull-ups and 10k pullups only (and changed ESP32 board after that, it was a hail mary, but I probably fried those UART pins since they're not 5v tolerant)

This is what I get in the log

<html>
<body>

20:52:32 | [D] | [MitsubishiHeatPump:241] | control - Was HeatPump updated? YES
-- | -- | -- | --
20:52:32 | [D] | [climate:378] | 'minisplit2' - Sending state:
20:52:32 | [D] | [climate:381] | Mode: OFF
20:52:32 | [D] | [climate:383] | Action: OFF
20:52:32 | [D] | [climate:386] | Fan Mode: OFF
20:52:32 | [D] | [climate:398] | Swing Mode: OFF
20:52:32 | [D] | [climate:401] | Current Temperature: nan°C
20:52:32 | [D] | [climate:407] | Target Temperature: nan°C

</body>
</html>

wBrhy2, 2023-06-02:
found how to access the chat, seems more appropriate

SwiCago, 2023-06-16:
@dmcc sent a pull that fixes chat link...

closed 2023-06-16 by SwiCago