SwiCago/HeatPump #202: Info for Ecodan Hydrotank

baldarn, 2023-09-14:
Hi! thanks for this amazing code!

I was wondering... 

I have a Ecodan hydrotank, that can be controlled with the mel cloud same as the splits.

this are some references:
(sorry for the italian, but I couldn't find those in english. the manual is multilingual :) )

https://climatizzazione.mitsubishielectric.it/it/prodotti-soluzioni/soluzioni-casa/pompe-di-calore/pompe-di-calore-packaged/hydrotank-packaged#col-product-info

https://climatizzazione.mitsubishielectric.it/sites/default/files/2023-05/Manuale%20di%20installazione%20Hydrotank%20e%20Hydrobox_compressed%20%281%29.pdf

How did you get the signals from the splits? can you give me an advice to integrate also the hydrotank?

thank you for your time ;)

SwiCago, 2023-09-20:
best way to do it is to dump packets between the unit and the device controlling it. 
if you have a melcloud you can do this one of two ways, use a logic analyzer or use an esp32  as a uart relay(connect melcloud to esp32 and esp32 to unit) have esp32 relay serial back and forth but also send packets to mqtt

closed 2023-09-20 by SwiCago

baldarn, 2023-09-20:
> best way to do it is to dump packets between the unit and the device controlling it. if you have a melcloud you can do this one of two ways, use a logic analyzer or use an esp32 as a uart [...]

thanks for the infos!

I'm very noob, can you link me some references to do that? I have some spare esp32 around ;)

SwiCago, 2023-09-23:
@baldarn , there are lots of examples of how to use serial on esp32, but I do know one person that did a esp32 relay so he could use MQTT and the MHK1. They posted theirninfo in chat. The main page of this repo has a link tonthe gitter chat. Inquire there.