SwiCago/HeatPump #207: Recent changes break bi-direction communication on MSZ-GL06NA

stewartadam, 2023-11-18:
I'm using this library via esphome-mitsubishiheatpump.

Commit cea90c5ed48d24a904835f8918bd88cbc84cb1be works as expected, but any commit later causes bi-directional serial communication to silently fail. Updates sent do nothing and no updates are received from the heat pump, so the current setpoint shows as NaN.

SwiCago, 2023-11-29:
@prashker , this relates to the commit you made, can you review pls

dzungpv, 2023-11-30:
> @SwiCago if I'm reading the above correctly, my commit works - it is **subsequent** commits breaking communication.
> 
> @stewartadam seems to indicate that problem is immediately after my commit, which would be https://github.com/SwiCago/HeatPump/commit/ce460c3bd226bea12643410cca9d631fc751b281 by @dzungpv 
> 

I am running for months without any issue. Both with esp8266 and esp32.

markalston, 2023-11-30:
@dzungpv  It looks like your issue is that you are only setting the values for rxPin and txPin inside the

`if (rx >= 0 && tx >= 0) {`

test.  

If on the initial call to connect, rx and tx are set to -1  then rxPin and txPin are not set but you use them on line 119.

`    _HardSerial->begin(bitrate, SERIAL_8E1, rxPin, txPin);`

Maybe you could  initialize them to -1 and add a test to check value before line 119 in case it is -1 and then fallback to the old 

`    _HardSerial->begin(bitrate, SERIAL_8E1);`

This is my guess to the issue anyway.

dzungpv, 2023-12-01:
> @dzungpv It looks like your issue is that you are only setting the values for rxPin and txPin inside the
>
> `if (rx >= 0 && tx >= 0) {`
>
> test.
>
> If on the initial call to connect, rx and tx are set to [...]

It is original code, I just store the custom RX TX pin for call in retry you can check change here: https://github.com/dzungpv/HeatPump/commit/ce460c3bd226bea12643410cca9d631fc751b281 and 4 of my units running fine since, 3 esp8266 and 1 esp32

dzungpv, 2023-12-01:
@stewartadam Can you check latest change?

SwiCago, 2023-12-04:
Thanks @dzungpv , @stewartadam , @markalston 
Great work of finding a solution for everyone fast...Pull has been merged.

closed 2023-12-04 by SwiCago

prashker, 2023-12-05:
Teamwork!