SwiCago/HeatPump #216: Question about remote temperature

sle118, 2024-03-28:
General behaviour question here.

How frequently should remote temperature be updated? And suppose we set a remote temperature of 21 while the set point is 22 via a mqtt integration and the mqtt server goes down for a while. Will the heat pump fall back to an internal reading, or will it just keep heating and eventually go above target?

Thank you!

mbbush, 2024-06-18:
I update mine once a minute. I think doing anything faster than a handful of times per minute would be excessive. 

There is no logic built in to the library to automatically fall back to the internal thermometer. So in the case you described the heat pump would continually call for heat until the mqtt server came back up. This is obviously not desirable behavior. 

I opened https://github.com/SwiCago/HeatPump/issues/190 a while back discussing whether to add this to the library, and the consensus was that it belongs at a higher level in the stack. Several implementations that use this library have some sort of timeout implemented in them.

sle118, 2024-07-11:
I originally wanted to implement my own solution, but abandoned the project as others were more advanced and had a faster pace. So I opted for https://github.com/muart-group/esphome, which the developers hope to merge as an official component of ESPHome some day.

SwiCago, 2024-09-25:
as @mbbush said, this should be done outside of lib. The lib does not know if you are using MQTT or a simple webpage to control the HP. So your primary code should have fails safes built in to detect if it is not longer getting the expected updates or lost connection to MQTT. Sending a 0 for remote temp will revert to internal control.

closed 2024-09-25 by SwiCago