SwiCago/HeatPump #223: No data

JamesDAdams, 2024-09-13:
Hi,

i plug my esp to my heatPump but only the uptime work...

any idea ?

[image]
[image]
[image]
[image]
[image]
[image]

SwiCago, 2024-09-25:
USe debug mode and look at the packets. If the initial packet to enable external control is sent over and over, then you have might not be receiving data back. To test if you are not getting data back, after init packet, send an ON command. If unit turns on, then your TX is fine and your RX is not. If the unit does not turn on, then your TX is bad. You may need a pullup resistor on one or both.

closed 2024-09-25 by SwiCago

apppie123, 2024-12-07:
You've plugged it into an Ecodan.. Swicago is not compatible with AirToWater pumps.