SwiCago/HeatPump #228: HJ50VA Support

fastbike, 2025-01-27:
> Hi All,
> 
> MITSUBISHI MSZ-HJ xxxx air conditioners which are normally not controllable because not having a CN105 connector, are finally.
> 
> After analyzing the technical documentation the control card has a CN105 which is not mounted and on the other side 2 0603 resistors are not mounted either. So I bought connectors and now I control my MSZ HJ50VA without any problem thanks to a WEMOS mini D1 with the Swicago firmware.
> 
> The resistors; I started cautiously with 470ohms, it did not work I replaced them with 100ohms and now it works perfectly. these resistors are in series in the Rx and Tx connections, I am almost sure that if we put a piece of wire so 0ohms it will work just as well.
> 
> The connector is an angled version mounted flat on the IC it is a JST 2mm 5 pins reference S05B-PASK-2 Bought on Aliexpress: [https://fr.aliexpress.com/item/1005005353133339.html?spm=a2g0o.order_list.order_list_main.5.17b05e5bUsiu9X&gatewayAdapt=glo2fra](url) it's not red but it still works lol
> 
> Hello/Bonjour!
> 
> @Docteur-No I am in your exact situation. I have a HJ50VA and the CN105 connector is not mounted/soldered.
> 
> May I reach you out to ask the modifications you did and how you managed to remove the PCB without breaking the grey support?Where did you find the documentation of the HJ50VA PCB?
> 
> Sorry for leveraging this thread, but it is the only way to connect to other people having the same model as mine. 

 _Originally posted by @lucabelluccini in [#13](https://github.com/SwiCago/HeatPump/issues/13#issuecomment-2614688433)_

fastbike, 2025-01-27:
There is a service manual available at https://planetaklimata.com.ua/instr/Mitsubishi_Electric/Mitsubishi_Electric_MSZ-HJ_VA_Service_Manual_Eng.pdf
Page 26 has the board layout, which shows CN105 as unpopulated. You will need to obtain the connector and follow the instructions given above.

[image]

lucabelluccini, 2025-01-27:
Hello @fastbike ! Thanks for forking out my request for help.

It's only the connector or I need to also take care of the R148, R144 resistors (~100 ohm) and JP3 & JP5 (to close)?

fastbike, 2025-01-27:
Sorry I can't help with that. However reading the original post it looks like some resistance in the serial line is a good idea, he tried 470R but found that was too high. If you do not have the skills required reach out to somebody closer who can help.

Docteur-No, 2025-01-27:
Hi Fastbike,

In fact the MITSUBISHI MSZ-HJ xxxx PCB has the CN105 location but it is not mounted on the board. By looking this board you can see than 2 resistors are unmounted too.

These resistors are in series in the Rx and Tx connections. if you have 2 x 100 Ohms you can mount them but o Ohm 2 (small wires) will do the job perfectly.

Do not forget above all you have to connect Rx of the MHZ to TX of the Wemos miniD1 and Tx of the MHZ to Rx of the Wemos. The Wemos is supplied by the 5V of the CN105.

It is very simple and work like a charm

lucabelluccini, 2025-01-27:
Thanks @Docteur-No 
To recap:
- R144 & R148 on the **power** board => 100 ohm each
   [image]
- CN105 => connector on the **control** board
    [image]

Docteur-No, 2025-01-27:
Yes that's right
If you look at the board you will see Rx & Tx from the Cn105 are open due to the 2 unmounted resistors. 100 Ohms works perfectly but I am sure zero Ohm works too

lucabelluccini, 2025-01-27:
I've ordered the connector. I'll try to implement this mid Feb. 
Thank you both for the help!
We can close this if you wish.

Docteur-No, 2025-01-27:
It’s a pitiy, have 9 unused !!!
Depend pon your country and shipping fee

Docteur-No, 2025-01-27:
You are right I am in France Béziers 
Do you want I send you one or two items

fastbike, 2025-01-27:
Awesome to see this issue resolved.

closed 2025-01-27 by fastbike

lucabelluccini, 2025-02-19:
For future reference:
- R1G3 & R1G2 would need a 100 ohm resistor each. I've short-circuited them and put the resistors on the wiring to the ESP
- The connector can be installed on the CN105

[image]

Wiring is (AC side >>> ESP32 side, in my case a ESP32 Wroom 32 - Dev kit v1):
- RED RX >>> TX GPIO17 or TXD2
- BLACK TX >>> RX GPIO16 or RXD2
- YELLOW 5V >>> VIN
- GREEN GND >>> GND