SwiCago/HeatPump #230: CN100 connector

osmaga2012, 2025-01-29:
Hello everyone. 

Could someone show an image of the project? 

My machine has a CN100 connector but I don't know if the diagram is the same as for the CN105. 

I want to mount it on an esp32. Could someone share the components of the project? 

thanks in advance. Greetings

fastbike, 2025-01-30:
Please provide more info: model, photos etc.

osmaga2012, 2025-01-30:
Thanks fastbike for the response. 

True, I didn't send any information about my team. These are the photos you can get and they are ducted air conditioning.

[image]

[image]

SwiCago, 2025-02-13:
Mapping(pin numbers) is shown on this page
https://github.com/SwiCago/HeatPump/wiki/Supported-models

closed 2025-02-13 by SwiCago

osmaga2012, 2025-06-03:
Hello again.
I've decided to take up this project again, and I have a question.
With an ESP32 development board, is the 5V to 3.3V voltage converter necessary? If I remember correctly, the ESP32 supported it....

Thanks in advance.
Best regards!